opencax/GSoC #96: Overlap removal

pca006132, 2025-01-26:
## Outline

Remove overlaps in meshes that contain self-intersection, assuming the mesh is a manifold.

## Details

Basically, a valid solid mesh should be both manifold and has no self-intersection. However, models from the internet may contain defects. This project is about coming up with an algorithm that removes self-intersections.

See https://github.com/elalish/manifold/issues/289 for details about ideas for the algorithm.

## Expected Outcome

Implementation of said algorithm.

## Future Possibilities 

## Project Properties

### Skills

- C++
- Graph data structure.
- Algorithms.

### Difficulty 

Hard.

### Size 

Long (350h)

### Additional Information

* Potential mentor(s): @elalish 
* Organization website: https://manifoldcad.org/
* Communication channels: https://github.com/elalish/manifold/discussions

vinayakjeet, 2025-02-19:
Hi @elalish and team,

I've reviewed the discussion in [#289](https://github.com/your-repo/issues/289) and the overall objectives for overlap removal ([#96](https://github.com/your-repo/issues/96)), and I'm excited to begin tackling this feature.🚀

vinayakjeet, 2025-02-19:
Hi @elalish @pca006132  and team,

I've reviewed the discussion in [#289](https://github.com/elalish/manifold/issues/289) and the overall objectives for overlap removal ([#96](https://github.com/opencax/GSoC/issues/96)), and I'm excited to begin tackling this feature.
🚀

vinayakjeet, 2025-02-19:
With respect to the discussion in [#289](https://github.com/your-repo/issues/289), how does the triangulator deal with polygons, like bowtie or star-shaped polygons with nested holes? Is this an area for improvement?

pca006132, 2025-02-20:
> With respect to the discussion in [#289](https://github.com/elalish/manifold/issues/289), how does the triangulator deal with polygons, like bowtie or star-shaped polygons with nested holes? Is this an area for improvement?

No, it works fine with these polygons. I think you can try to look at the regular triangulation algorithms first, understand how they work with non-convex polygons and polygons with holes, and then come back to this.

elalish, 2025-02-20:
Here's some info on our triangulator: https://github.com/elalish/manifold/wiki/Manifold-Library#polygon-triangulation

What's your background? Have you written computational geometry algorithms before?

vinayakjeet, 2025-02-23:
I've explored curve approximations but haven't written full geometry algorithms. I'll study triangulation methods and Manifold's approach—thanks for the reference!

vinayakjeet, 2025-02-23:
Do you have an IRC channel where I can ask questions?

pca006132, 2025-02-24:
There is an unofficial discord: https://discord.gg/22kHZVpKBP, manifold channel.

elalish, 2025-02-24:
You're also welcome to ask questions here or make a github discussion thread - I haven't gotten in the habit of checking discord yet, though I can if necessary.

vinayakjeet, 2025-02-24:
No issues, I'll ask here if needed.

wasup-yash, 2025-02-24:
Hey, This feature seems fascinating to work on, I read the discussion on overlap removal and would be starting to draft a proposal with a formulated structure on how to go about this, Need to go through the methods too, and about their implementation meanwhile :) 
also, can you inviteme again ? cause the above invite of discord seems to be expired.

pca006132, 2025-02-25:
> Hey, This feature seems fascinating to work on, I read the discussion on overlap removal and would be starting to draft a proposal with a formulated structure on how to go about this, Need to go [...]

Weird, it seems that it is not expired:

[image]

Anyway, here is another link that hopefully is not yet expired: https://discord.gg/6PB7qZ4HC7

wasup-yash, 2025-02-25:
> Weird, it seems that it is not expired:
>
> [image]
>
> Anyway, here is another link that hopefully is not yet expired: https://discord.gg/6PB7qZ4HC7

thank you, Joined

pca006132, 2025-04-20:
@wasup-yash @vinayakjeet btw, do you have any questions about the algorithm we sketched here and in the associated issue https://github.com/elalish/manifold/issues/289?

vinayakjeet, 2025-04-23:
@pca006132

> Thanks for checking in! I don’t have the resources or experience right now to fully understand the algorithm end to end, but I’m definitely interested and would love to explore it more — even outside of GSoC, if I don’t get selected 🚀

elalish, 2025-04-24:
Do you have any questions about it? This is a great place to ask.