opencax/GSoC #97: Manifoldness repair

pca006132, 2025-01-26:
## Outline

Repair triangle soup that are not manifold.

## Details

Basically, a valid solid mesh should be both manifold and has no self-intersection. However, models from the internet may contain defects. This project is about coming up with an algorithm that converts and repair a triangle soup into a manifold mesh.

This will contain a lot of heuristics, basically what we need is:

- Stitching faces together, and maybe join faces that are close enough.
- Fill holes.
- Duplicate vertices and edges such that the result is a manifold in terms of connectivity.

## Expected Outcome

Implementation of said algorithm.

## Future Possibilities 

## Project Properties

### Skills

- C++
- Graph data structure.
- Algorithms.

### Difficulty 

Hard.

### Size 

Long (350h)

### Additional Information

* Potential mentor(s): @elalish @pca006132 
* Organization website: https://manifoldcad.org/
* Communication channels: https://github.com/elalish/manifold/discussions

elalish, 2025-01-29:
We have a starter algorithm for this: [Merge()](https://manifoldcad.org/docs/html/structmanifold_1_1_mesh_g_l_p.html#a2078d075bef279041b67769e4197c254) that only merges nearby edges. It's conservative and gives up if the mesh is not close to manifold to begin with. Still, it's a reasonable place to start. 

We do not expect a result free of self-intersections for this project - that's a separate project. In general, a non-manifold mesh can require a lot of heuristic additional surfaces, so part of this project is artistic in nature: what additions make the most sense and look reasonable?

Abhyudaya30Singh, 2025-02-19:
Hi @elalish @pca006132 , I'm Abhyudaya Singh, and I would love to contribute to this issue! I notice the Merge() function assists in stitching together neighbouring edges. What are its existing constraints when working with non-manifold meshes, specifically situations where the merge vectors were compromised? Are there known failure modes or edge cases that need special treatment?

pca006132, 2025-02-20:
I think you can check the definition of [manifoldness](https://github.com/elalish/manifold/wiki/Manifold-Library#manifoldness) first. By non-manifold, we typically mean meshes with holes, as well as meshes that are non-manifold geometrically (due to shared vertices/edges). I am not sure what you mean by "the merge vectors were compromised".

elalish, 2025-02-20:
Generally the merge vectors are not so much "compromised" as missing entirely. `Merge()` can only stitch together verts within tolerance of each other - it simply hopes this is enough to make a manifold. For any larger gaps or holes, a more serious bunch of heuristics are necessary. Our triangulator may help with hole filling, but it may need some help with significantly non-planar ones. Perhaps something like our `CreateFaces` function could first break the holes into more nearly planar patches? Anyway, there are lots of possibilities. First step is to go find a set of popular meshes that aren't manifold (which is most of them), so we can test.

elalish, 2025-02-20:
I'd recommend looking at art / video game style assets and 3D scans - they often have much worse manifoldness than e.g. 3D printing meshes.

Abhyudaya30Singh, 2025-02-25:
I have collected a bunch of 3D scans, and I'm attempting to use Merge(). But I'm not entirely certain if I'm doing it correctly. How should I test them? And how do I determine if it actually worked? @elalish

pca006132, 2025-02-26:
You can look at how our tests uses this `Merge` function, e.g. https://github.com/elalish/manifold/blob/cba9e35bb1ab982c68d5d1e7a88ce90f7eead91f/test/manifold_test.cpp#L535-L556. If it doesn't work, the status will contain some error.

Abhyudaya30Singh, 2025-03-23:
I've implemented an OBJ file analysis system using Manifold, featuring mesh metrics analysis, boolean operations, and robust error handling with comprehensive tests. @pca006132

Abhyudaya30Singh, 2025-03-23:
What repair strategies would you recommend for non-manifold meshes? Should I prioritise extending the OBJ parser or implementing repair tools next?

Abhyudaya30Singh, 2025-03-23:
https://github.com/Abhyudaya30Singh/mesh-analysis-tool

pca006132, 2025-03-24:
I don't think it is a good idea to describe this as having "robust error handling with comprehensive tests" when you only called manifold, dumped some of the statistics out, and have 1 test case related to the use of the `Merge` function. Also, I think you have some misunderstanding regarding how `Merge` works: It is used for fixing non-manifold mesh, there is nothing to fix for manifold meshes.

Can you explain this code: https://github.com/Abhyudaya30Singh/mesh-analysis-tool/blob/ae2119eb785ad481f47a4bc4753c890a1bcb2073/test/test_obj_analysis.cpp#L50-L82

How are they related to `Merge`? Why did you write these tests?

Also, you mentioned you got some 3D scans and used `Merge` on them. Where are they? What results did you get? Your repo only contains cubes.