opencax/GSoC #98: [Placeholder] Authoring interface for IFC4.3 alignment geometry in Bonsai

aothms, 2025-01-29:
## Outline

## Details

## Expected Outcome

## Future Possibilities 

## Project Properties

### Skills

### Difficulty 

### Size 

### Additional Information

* Potential mentor(s): [NAME](https://)
* Organization website: https://
* Communication channels: https://******.zulipchat.com

RickBrice, 2025-02-10:
> WARNING: This email originated from outside of WSDOT. Please use caution with links and attachments.
>
> Draft GSoC proposal on IfcOpenShell wiki<https://github.com/IfcOpenShell/IfcOpenShell/wiki/Infrastructure-Authoring-%E2%80%90-Design-and-Coordination#gsoc-proposal>

Thanks Scott – I made a couple of tweaks. Added (optionally) transition curves to the scope and added some technical references.

I have a meeting later this week with some professors from my local civil engineering school. What does the process look like going forward? I would guess something like:

  1.  Agree on the GSoC posting
  2.  Apply to be a GSoC mentoring organization (probably handled by Thomas and Dion)
  3.  Wait for acceptance
  4.  Get people to apply to be GSoC contributors
  5.  Review applications, select contributors, and execute the project.

Rick

Richard Brice, PE
Bridge Design Technology Unit Manager
WSDOT Bridge and Structures Office
360-705-7174

RickBrice, 2025-02-13:
[image]

### Outline

Industry Foundation Classes (IFC) offer the ability for rich information exchanges between modeling, analysis, planning, and other software tools in the Architecture, Engineering, and Construction (AEC) industry.  Specifically, the latest release of IFC (version 4.3, also referred to as IFC4X3) adds linear referencing via alignment modeling, which is core to describing the construction and maintenance of infrastructure assets such as roads, bridges, and railways.

### Details

Alignment import (read) capabilities have been added to IfcOpenShell and the Bonsai add-in for Blender. They have reached a state of maturity such that the next logic step is to enable alignment authoring (write) capabilities.  

### Expected Outcome

Alignment authoring will take place in Blender via the Bonsai add-in.  A user-focused workflow has been developed and documented, along with preliminary user interface mockups. This project would add alignment authoring capabilities via new panels and other items within Blender.  The ifcopenshell.api namespace will also need to be enhanced incrementally to support the new user interface tools.

### Project Properties

#### Skills

Understanding and general working knowledge of python.

#### Difficulty

Medium

#### Size

##### Medium (175 h)

The participant focuses on authoring horizontal alignments via the PI method.  This could be via interactive icons or primarily through a table-based interface.  The user would need to be able to add, edit, and remove PI (point of intersection) points.  Additionally the user would need to be able to adjust the radius that corresponds to each PI point. Though not strictly required for this project, the authoring tool would also enable definition and editing of entry and exit transition curve type (clothoid, sine spiral, polynomial spiral, etc.) and length.

##### Long (350 h)

PI-based alignment would be added for vertical and cant as well.  A basic corridor modeling UI tools would be implemented to allow for sweeping geometry (open or closed profile) along an alignment curve to generate 3D linear geometry via IfcSectionedSolidHorizontal and related IFC entities.

#### Additional information

Mentors: Rick Brice @RickBrice & Scott Lecher @civilx64

Organization website: https://ifcopenshell.org

Communication channels: https://github.com/IfcOpenShell/IfcOpenShell/discussions

Technical resources:

https://docs.bonsaibim.org/guides/development/index.html

[Blender 4.3: Precise Modeling for Architecture, Engineering, and 3D Printing](https://a.co/d/cdryOQQ)

[Python Scripting in Blender](https://a.co/d/3us1PuT)

AceHunterr, 2025-03-30:
Greetings @aothms , I am Mehul Aggarwal and I am interested in contributing for this project during GSOC if it is still available?

As per my understanding of the project the work that is supposedly to be done includes:

- Building a User Interface: Developing new panels and tools in Blender so that users can easily add, modify, or remove PI points.
- Editing Alignment Data: Making it simple to tweak how curves and straight segments connect.
- Integration with Existing Tools: Enhancing the ifcopenshell.api (the programming interface for IFC operations) to support these new UI features, ensuring that the alignment data is correctly written to IFC files.
 

In short, developing Python code to create a user-friendly tool in Blender ensuring IFC standards.

Apart from the mentioned resources are there any other walkthroughs that I need to study or any specific section that I need to get covered in my GSOC proposal?

aothms, 2025-03-31:
Hi @AceHunterr welcome, note that @RickBrice @civilx64 are the mentors on this project, so I'm speaking for them here.. Your summary is somewhat accurate: 

The context we provided is mostly to understand the various layers involved in IfcOpenShell:

- fileformat, schema, geometry interpretation (C++)
- python bindings 
- high level ifcopenshell.api (python)
- user interface (blender)

So this project is mostly in the bottom 2 layers (but also has a geometric component of course). Intention is to develop most functionality in the API layer so that people can also make use of the functionality in a programmatic context, not only from the UI.

For the project proposal, focus on your prior experience, motivation, planning and goals within this project. We have the requirement that every project proposal is accompanied with a Pull Request to the ifcopenshell repo to demonstrate some familiarity with the code and general programming principles.

I'm sure @RickBrice @civilx64 have some good recommendations for the civil engineering background. Some random resources (fairly challenging) to see how we have the IFC schema standard, implementations for those in C++, example models https://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcClothoid.htm https://github.com/IfcOpenShell/IfcOpenShell/blob/cba6161043dd9bf79a9d058beedd8f6b6607037d/src/ifcgeom/mapping/IfcCurveSegment.cpp#L528 https://github.com/buildingSMART/IFC4.x-IF/blob/main/IFC-files/ACCA/Bridge-for-MINnD/Viadotto%20Acerno.zip

Best of luck, looking forward to seeing your proposal!

AceHunterr, 2025-04-01:
Thanks for the guidance @aothms ... I will take care of the points that you have mentioned here. I have 2 specific doubts here:

First , you mentioned that I need to have some a Pull Request in the IFC OpenShell repository... so should I solve any small issue or like make any particular change to demonstrate... what could I make the PR about?

Second, is there any particular format for the proposal that is needed to be submitted?

RickBrice, 2025-04-01:
@AceHunterr The requirements for the pull request are a bit vague. They are mentioned in item 7 of the application checklist. https://opencax.github.io/gsoc_checklist.html. Maybe try to add some documentation or an example. Demonstrate that you are familiar with the project.

Here you will find general guidance for writing proposals and applying. https://opensource.googleblog.com/2025/03/google-summer-of-code-2025-contributor.html

The application checklist has a link to the specific application format. https://opencax.github.io/gsoc_guidelines.html

AceHunterr, 2025-04-01:
I see @RickBrice ... thank you for the resources and I'll surely get the things required done asap.

Moult, 2025-04-01:
@AceHunterr also feel free to jump into the IRC/Matrix chatroom https://osarch.org/chat to say hello to the community - we can help guide around the Blender dev setup :)

civilx64, 2025-04-22:
@AceHunterr just a reminder that we would like to see some code contributions - ideally a pull request.  Here is a suggestion: https://github.com/IfcOpenShell/IfcOpenShell/issues/6584. Feel free to follow up on that issue and tag the author with any questions.

AceHunterr, 2025-04-22:
Greetings @civilx64  and I deeply apologise for such length of time taking... I have been caught up with my final semester examinations and multiple project evaluations during this month and for some part of the next month as well therefore was not able to dedicate my time currently in this... I'll try to make the PR as soon as possible.
Apologies again.

civilx64, 2025-04-22:
No problem. Good luck!