opencax/GSoC #100: Integrated language help feature in OpenSCAD

t-paul, 2025-02-09:
## Outline

Add more interactive help features for built-in functions and modules. Right now there's already a nice summary of parameters linked as [cheat sheet](http://www.openscad.org/cheatsheet/). Scope of this projects would be to use this information in extended form and make it available in a more direct way in the editor.

## Details

* Convert the cheat sheet information into machine readable format
* Find a way to generate the existing HTML format based on the core data
* Add context help to editor giving help for built-in functions and modules, e.g. by adding formatted help output to the console window, including the links to further documentation like the language manual on Wikibooks

## Expected Outcome

Cheat sheet is integrated into the application and additional context help for built-in functions and modules is available.

## Project Properties

### Skills

* Programming language is C++
* GUI programming with the Qt framework

### Difficulty 

Easy

### Size 

Medium (175h)

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://web.libera.chat/?channel=#openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://web.libera.chat/?channel=#openscad): teepee)
* Organization website: https://www.openscad.org/

guptushar27, 2025-02-10:
I want to work on this issue. Please Assign this to me.
Thanks

t-paul, 2025-02-10:
@guptushar27 Those projects are planned for GSoC, for how that process works, see https://summerofcode.withgoogle.com/how-it-works and https://opencax.github.io/ - If you want discuss, join the IRC channel, link above.

siyangwu358, 2025-02-18:
How can I join your channel to discuss with you.

mpark341234567, 2025-03-30:
Hi Mentors,

My name is Michael, and I am a second-year community college student. I am interested in this project for GSoC, but unfortunately, I do not know GUI programming. Can I learn GUI before the project starts? Also, I would like to have more information on this project. Thanks!

t-paul, 2025-03-30:
@mpark341234567 As the name Google Summer of *Code* indicates, it's focus is about coding. If you have some basic knowledge of C++ and only some specifics like the GUI library are new that could be ok, there is also still some time till the project would actually start. Going without *any* of the skills listed is probably not a good idea, in that case it would be advisable finding a different project with better match.

mpark341234567, 2025-03-30:
Thank you for the clarity! I am currently taking the second-level course on C++ programming, which includes some OOP topics like inheritance and polymorphism. The GUI library, however, is completely new to me. Am I suitable for this project, or should I look for a better match?

t-paul, 2025-03-31:
At this point, it's mostly for you to judge if you feel ready to take on the topic. Maybe take a couple of hours and get the source code and build it. Have a look at the existing code and the running application - see if you have a basic idea what the project would want to achieve (talk to us on IRC for help with that!). If you are ready to write a detailed project proposal based on the information you have collected, and see some direction to start on the code side, then go with it.