opencax/GSoC #101: Development and Build Support for Native Windows

kintel, 2025-02-10:
## Outline

OpenSCAD is multi-platform desktop application, with official support for Windows, macOS and Linux, and unoffical support for various other Unix-like OSes like FreeBSD.

While OpenSCAD does support Windows, Windows development and build setup is a bit suboptimal:
* Windows binaries are built on Linux using the MXE cross compilation environment (https://mxe.cc). [Build setup](https://github.com/openscad/openscad/blob/4e9d40a74e7ecb9fa2e5744df59fd5a633e9ae28/.circleci/config.yml#L7)
* Windows native builds, including tests are done on msys2 (https://www.msys2.org) [Build setup](https://github.com/openscad/openscad/blob/4e9d40a74e7ecb9fa2e5744df59fd5a633e9ae28/.github/workflows/windows.yml#L22)

Goal of this project is to improve Windows support for native Windows development.

## Details

* Adjust OpenSCAD source code as needed to build OpenSCAD using MSVC
* Adjust the CMake build system as needed to support MSVC
* Find a way of managing building (and packaging if needed) 3rd party dependencies. OpenSCAD depends on a number of 3rd party packages (e.g. Qt, QScintilla2, CGAL, Manifold, GMP, MPFR, boost, OpenCSG, GLEW, Eigen, glib2, fontconfig, freetype2, harfbuzz, libzip, Bison, Flex, double-conversion). Not all of these have great Windows build support.
* Integrate library building/packaging into our Continuous Integration framework (e.g. GitHub Actions or CircleCI).
* Build OpenSCAD on MSVC + run tests natively for every PR, as we do for other build environments. This to make sure contributions don't fall out of maintenance.
* Establish support for debugging OpenSCAD in the MSVC debugger, and documentation on how to set that up, if necessary
* Write/update documentation on how to establish a native WIndows development environment
* Consider switching the official OpenSCAD Windows build to use MSVC.

Prior work: https://github.com/openscad/openscad/issues/4976

## Expected Outcome

A native MSVC build environment for OpenSCAD is reasonable easy to set up, and such an environment is continuously integrated.

## Project Properties

### Skills

* Good understanding of the Windows OS and native components (DLLs, executables) on a low enough level to be able to debug odd behaviors.
* Experience using MSVC and native Windows development for C++ projects
* Good understanding of how 3rd party libraries are built and distributed.
* Experience with or interest in acquiring skills using CMake and writing custom CMake configs and macros
* Experience with GitHub CI or CircleCI is a bonus

### Difficulty

medium

### Size

long (350h)

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://web.libera.chat/?channel=#openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://web.libera.chat/?channel=#openscad): teepee)
   * Note: None of the mentors have relevant Windows skills, but have excellent understanding on how all the relevant technologies work and how they are integrated with OpenSCAD. It's important that the candidate is able to acquire any necessary Windows-specific skills independently.
* Organization website: https://www.openscad.org/

Vai-Man, 2025-03-18:
Hello @kintel, @teepee,

I’m Vaibhav, a CSE student, with experience in C++, CMake, and software build systems. I have contributed to open-source projects and worked with GitHub Actions for CI/CD. I’m interested in improving OpenSCAD’s native Windows development support.

From what I understand, OpenSCAD currently relies on MXE for cross-compilation and MSYS2 for native builds, but the setup isn’t fully optimized for Windows development. The project aims to enable an MSVC-based build system, integrate it into CI, and improve dependency management.

I’d like to understand the current blockers in transitioning to MSVC. Are there specific third-party dependencies that pose major compatibility challenges? Also, is there an existing partially functional MSVC setup, or would this require building from scratch?

Before drafting my proposal, are there any specific areas I should explore further, such as prior build attempts or debugging workflow improvements?

Looking forward to your insights!

kintel, 2025-03-26:
> I’d like to understand the current blockers in transitioning to MSVC. Are there specific third-party dependencies that pose major compatibility challenges? Also, is there an existing partially functional MSVC setup, or would this require building from scratch?
> 

The Details section above lists most topics needing addressing, and it includes a link to prior work.
None of the existing devs are familiar with Windows or have readily available MSVC setups, so we're really relying on someone to teach us how that all works. The main blocker is bringing the Windows native dev skills to the table.

> Before drafting my proposal, are there any specific areas I should explore further, such as prior build attempts or debugging workflow improvements?
> 
I would suggest starting by building and running OpenSCAD using one of the existing build setups for Window, then engage with the community on IRC to demonstrate your abilities to communicate and work with the tooling in questions.

As with most applications, we rely heavily on assessing candidates based on communication done via IRC and GitHub and, in lieu of having an interview process, we rely on demonstrated ability to engage with our codebase and community to demonstrate needed skills. Cold applications are very rarely successful.

closed 2026-02-03 by t-paul