opencax/GSoC #102: Improving the k-File to BRL-CAD Converter

drossberg, 2025-02-13:
## Outline

In the past years, we put some effort in the development of a LS-DYNA keyword file to BRL-CAD converter. Although we made great progress there, we still can't convert every k-file to g, i.e. the native BRL-CAD format. The goal of this project is to increase the number of covertable k-files. 

## Details

The sources of the current k-file to BRL-CAD converter can be found in the brlcad repository at [src/conv/g](https://github.com/BRL-CAD/brlcad/tree/main/src/conv/k-g). You have to compile BRL-CAD from its sources to work on this project and see the effects of your changes.

Examples of k-files, which cannot be converted with k-g, can be found here: [THUMS](https://www.toyota.co.jp/thums/) You can however use your own examples.

## Expected Outcome

We expect an improved k-g LS-DYNA keyword file to BRL-CAD converting program as the outcome from this project.

## Project Properties

### Skills

* C/C++
* LS-DYNA or a similar FE solver software, which can read k-files (needed as reference for how the geometry should look like)

### Difficulty

medium

### Size

This project could have any size, short (90h), medium (175h) or long (350h), depending on the amount of functionality you want to add.

### Additional Information

* Potential mentor(s):
[Ali Haydar](mailto:devs@brlcad.org)
[Daniel Rossberg](mailto:devs@brlcad.org)
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

MihirGore23, 2025-03-01:
Hello @drossberg 

I’m Mihir Gore, a junior at VJTI Mumbai, with experience in C++, Python, and computational geometry. I was a GSoC 2024 contributor with JdeRobot, and I’m interested in contributing to this task.

From what I understand, LS-DYNA k-files primarily store finite element (FE) mesh data, while BRL-CAD relies on constructive solid geometry (CSG). The conversion process seems to involve mapping FE-based geometries to CSG primitives, which can be non-trivial, especially for meshed surfaces. I wanted to ask—what are the major failure cases in the current converter? Are there specific LS-DYNA element types that don’t translate well into BRL-CAD’s format? Understanding these limitations would help me focus on the key areas that need improvement. Also, are there any starter tasks that I need to complete before getting hands-on with contributing and proposal drafting?

drossberg, 2025-03-02:
Hi @MihirGore23, thanks for your interest in contributing to BRL-CAD.

Lets start with your last point: You should familiarize yourself with BRL-CAD. I.e., install the programs and go through the tutorial. Also, you should join our community chat https://brlcad.zulipchat.com. There, you can ask questions, e.g. **Louis Googl** about k-files. He can tell you about the issues of the current converter (mainly unhandled formats and keys in the k-files).

AliHaider93, 2025-03-02:
Hello @MihirGore23, as @drossberg mentioned, you can join us on the Zulip chat for further discussions. I also recommend installing LS-PrePost, the pre- and post-processor for the LS-DYNA solver. Familiarizing yourself with it will help you visualize models, which can be crucial for debugging.

AliHaider93, 2026-01-16:
Hi @lvwln, you can also discuss your idea for the pull request on the zulip channel. https://brlcad.zulipchat.com/#narrow/channel/111975-Google-Summer-of-Code/topic/k-g/with/547140599