opencax/GSoC #103: Support for arbitrary parametric objects

Moult, 2025-03-06:
## Outline

3D geometry is currently defined by a data schema known as IFC. This project is to build an interface and feature in Blender to link an object's geometry to any external geometry source (e.g. Blend file, OBJ, STL, etc).

This means people can use their preferred modeling tool or parametric object generation script and the result can still be synchronised with the building model.

## Expected Outcome

A new interface panel in Blender, and an established convention on how to read and sync these external data sources.

### Skills

Python, some Blender

### Difficulty 

Easy to medium

### Size 

Medium

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://osarch.org/chat

Akshika-Munshi, 2025-03-08:
@Moult Hello, I am interested in working on this project for GSoC. Is it still available?

Moult, 2025-03-08:
Yep

Dion Moult

hunter-nguyen, 2025-03-13:
Hello @Moult . I'm also interested in working on this project for GSoC. I have experience working with and creating scripts with 3D designs and reached out to the @brlcad.org email as well. Would love to get started!

AbhiNammu, 2025-03-27:
Hello@moult.I'm also interested in this project and would like to work on it for GSoC.Could you please guide me?

Moult, 2025-03-27:
G'day, the best thing to do is to reach out in osarch.org/chat, and have the software already installed and started playing around. Having a draft proposal would also really help.

Abhiram0407, 2025-03-27:
@Moult would you please guide me to write a proposal as i just want to consult you and take your guidance for writing the proposal as with mentor help the proposal would look more formal

DhanushA1307, 2025-04-04:
@Moult Hello, I am interested in working on this project for GSoC. Is it still available?